Markiv9999/STM32NUCLEOBUS #1: I2C sub-classes

StefanoSperetta, 2024-06-15:
The I2C classes are completely separate and this forces code to differentiate between each class. This leads to a lot of boilerplate code to be needed (and maintained), which might easily become a source for errors.

I would suggest to define the I2C_DMA and I2C_IT classes as children of I2C: this will create an easy hierarchy and it will allow to overload the methods such that every class using I2C can be used with either of its children, as the function calls are all identical.

This inheritance will simplify the code and reduce the boiler plate, making the code much more portable. [This comment](https://github.com/Markiv9999/STM32NUCLEOBUS/blob/2af3cfbe3922883b277fc7a064beb996e60660de/Core/Inc/TMP100.h#L33-L35) sums it all up very well: the TMP100 code should be customized depending on which I2C class is used (IT, DMA, etc..) making it non-standard and requiring customization. This strategy should be avoided as leads to a lot more code to maintain and potentially more bugs to appear in the near future.

StefanoSperetta, 2024-06-16:
The [class definition](https://github.com/Markiv9999/STM32NUCLEOBUS/blob/487cbe00801637df432a073dfcdcd8f17b7b4b15/Core/Inc/I2C_IT.h#L24) should become:

``` 
class I2C_IT : public I2C 
{ 
...
```

This will make sure the I2C_IT class (and all other derived classes from I2C) derive from it.
All common functions (if any, that are 100% identical) can be in the I2C class and all specific ones should then be in the I2C_IT class.

It is important to also remove [copy constructors](https://github.com/Markiv9999/STM32NUCLEOBUS/blob/487cbe00801637df432a073dfcdcd8f17b7b4b15/Core/Inc/I2C.h#L36-L37): there should be one and only one I2C (or derived) class per each I2C peripheral. Having multiple ones creates problems as the hardware status is mimicked only by one of the I2C objects and this will create inconsistencies in the hardware behavior.

Markiv9999, 2024-06-16:
I made an attempt today to set up the class inheritance. There is a weird issue where I2C_DMA is only able to inherit I2C when both are in the same header file. If they are in separate files, which I wanted to keep because it would help clarity, the compiler shows the following error:

> expected class-name before '{' token	I2C_DMA.h	/CommunicationBus/Core/Inc	line 24	C/C++ Problem

I made sure the header file of I2C is called in I2C_DMA.

But mainly wanted to ask, if it is not an issue, we could just keep all the classes in the same header file.

Markiv9999, 2024-06-18:
The only workaround that seemed to work for this issue is putting all the classes in the same file. I also tried forward declaring the classes but that didn't seem to fix the issue.  

Have included comments such as // DMA***********************************************DMA to indicate seperation in the cpp files.

Will leave this as an open issue for now will come back to it later if needed

closed 2024-06-26 by Markiv9999