Markiv9999/STM32NUCLEOBUS #3: C++ code inside a C code block

StefanoSperetta, 2024-06-15:
C++ code (see [this](https://github.com/Markiv9999/STM32NUCLEOBUS/blob/2af3cfbe3922883b277fc7a064beb996e60660de/Core/Inc/TMP100.h#L29)) should not be placed inside a C block (see [this](https://github.com/Markiv9999/STM32NUCLEOBUS/blob/2af3cfbe3922883b277fc7a064beb996e60660de/Core/Inc/TMP100.h#L17-L19)) as the class statement is not supported by a C compiler. This works because the project definition is weird in the sense that the compiler is instructed to support C++ code in C code blocks.

This is in principle an error coming from a project configuration: the configuration issue should be solved and these C code blocks (also including [this](https://github.com/Markiv9999/STM32NUCLEOBUS/blob/2af3cfbe3922883b277fc7a064beb996e60660de/Core/Inc/TMP100.h#L62-L64)) should be removed.

Markiv9999, 2024-06-15:
Have removed these from the code. Code is working without them as of now

StefanoSperetta, 2024-06-16:
Great! There is only one extern C piece of code left in [main.h](https://github.com/Markiv9999/STM32NUCLEOBUS/blob/487cbe00801637df432a073dfcdcd8f17b7b4b15/Core/Inc/main.h#L27-L29) but do not touch it. I have tried to understand why this is necessary and I think it has to do with the way STM32CubeIDE deal with automatically generated code. 

I really do not like this way of splitting [initialization code](https://github.com/Markiv9999/STM32NUCLEOBUS/blob/main/Core/Src/stm32l4xx_hal_msp.c) for certain peripherals and [normal execution code](https://github.com/Markiv9999/STM32NUCLEOBUS/blob/main/Core/Src/main.cpp) but this is tomorrow's problem.

Closing the issue now

closed 2024-06-16 by StefanoSperetta