Markiv9999/STM32NUCLEOBUS #4: HAL_I2C_Master_Transmit_IT keeps calling interrupt handler

Markiv9999, 2024-06-15:
When trying to transmit data in blocking mode, the I2C events interrupt keeps getting triggered to form an infinite loop.

Testing shows the issue does not arise from the inputs to the HAL command (tried many variations) or from the definition of the I2C class (tried the same code without the class in main).

Could indicate a possible issue with the HAL drivers, which do seem to defined ok.

Other interrupts such as GPIO Push button interrupts are still able to trigger if priority is defined appropriately.

StefanoSperetta, 2024-06-15:
I di not see any transmit done interrupt handler in the code. I guess you should define a function to be called when the transmission of one byte (or block of bytes) completes. This function loads the new bytes to send or simply clears the interrupt flag, ensuring the event is only triggered once.

Markiv9999, 2024-06-16:
There is an automatically generated interrupt handler in stm32l4xx_it.cpp (the event interrupt handler) and this does get called (Also the debugger is not able to follow this call when running the code line by line, but a print statement did get printed and this is the one that loops). This event handler calls the default HAL I2C event handler, which seems to be the one which fails to set the flags.

Read online that it could be an issue with the hal code version, tried downgrading to 1.16.0 (currently latest is 1.18.0), and that did not work.

Also tried changing the I2C channel to I2C2, and a few other things, and the issues remains.

The transfer complete interrupts are according to the documentation,
`HAL_I2C_MasterTxCpltCallback(I2C_HandleTypeDef *hi2c)`
and
`HAL_I2C_MasterRxCpltCallback(I2C_HandleTypeDef *hi2c)`
These functions are defined weak in the HAL driver code and are supposed to be called automatically according to the documentation. Thought defining them could be the answer.
So I redefined them in the main and added a print statement (the commented code is still there in the NucleoTMP100). Still had the same issue and the print was not triggered in the running code. 

Likely as you said it is the interrupt handler, will look into it more and update.

Markiv9999, 2024-06-22:
The issues is solved!!!!!!

The initialization of the Interrupt was ok (but stupidly complicated main->initI2C->the init function in Hal code which then called-> MspInit).

Event handler is passed to a pointer(XferISR) inside HAL_MASTER_TRANSMIT_IT and similar functions and is defined in stm32l4xx_it.h .

This issue was that in I2C.h, the I2C_HandleTypedef hi2c1 was defined as a copy instead of a reference (I2C_HandleTypeDef &hi2c1). The event handler was defined to operate on the hi2c1 defined in main (the original which was not set up for transmission) while all the configuration for transmission was made to the hi2c1 copy inside the I2C class. 

This was not an issue for the blocking function, as no code outside of I2C ever messed with the I2C_HandleTypeDef object inside.

Once we set this the variable to a reference, this problem was solved.

closed 2024-06-22 by Markiv9999