Markiv9999/STM32NUCLEOBUS #7: TMP100 user-defined configuration

StefanoSperetta, 2024-06-16:
In TMO100.h there is a function that accepts a configuration parameter (which specifies how the internal configuration of the IC works and how the temperature conversion also works). This parameter should not be left to the user to configure.

The user could set a different value than 0x60 [here](https://github.com/Markiv9999/STM32NUCLEOBUS/blob/487cbe00801637df432a073dfcdcd8f17b7b4b15/Core/Src/main.cpp#L111-L114) and the code would stop working. Since only 0x60 works, this should not be user selectable, but fixed in the TMP100 class.

In the [original TMP100 code](https://github.com/DelfiSpace/TMP100/blob/cf62dae5bc2167ffc38d400f358f886eb3662625/TMP100.cpp#L32), the function was called init as it initialized the IC and it provides no configurable parameter to the user (as only one option would work). Moreover, 0x60 is not stated anywhere but the bit names (by means of [#define statements](https://github.com/DelfiSpace/TMP100/blob/cf62dae5bc2167ffc38d400f358f886eb3662625/TMP100.h#L21-L56)) are provided, making the code much easier to understand.

I also think the name `init` (or `configure` or similar) would be much clearer for a user to understand.

Markiv9999, 2024-06-16:
Modified this value to be static. Had kept it variable before because TMP100 had a couple of interesting setting such as the alert high and low temp alerts which might be useful in some situations.

Markiv9999, 2024-06-18:
Closing the issues as the changes are now included in the code

closed 2024-06-18 by Markiv9999