Markiv9999/STM32NUCLEOBUS #8: Remove dependency on stm32_hal in TMP100

StefanoSperetta, 2024-06-16:
TMP100 depends on the STM32_hal only due to the integer definition (uint8_t, etc..) but the integer definition comes from stdint.h (nested through the different hal .h files.

[This line](https://github.com/Markiv9999/STM32NUCLEOBUS/blob/487cbe00801637df432a073dfcdcd8f17b7b4b15/Core/Inc/TMP100.h#L20) should thus become:

```
#include <stdint.h>
```
 and then the TMP100 code will be completely independent from the any STM32 code and fully portable. Its only other dependency will then be I2C, which makes sense. stdint is a standard C/C++ library present in any compiler so it limits the risk of future compatibility issues.

Markiv9999, 2024-06-16:
Done!! Have modified the include.

Markiv9999, 2024-06-16:
Will close this issue

closed 2024-06-16 by Markiv9999