Markiv9999/STM32NUCLEOBUS #24: I2C GPIO initialization not in I2C class

StefanoSperetta, 2024-07-09:
The I2C class initialized the I2C peripherals but it does not initialize the I2C pins: I would move this code also in the I2C class as you currently have halp of the I2C system initialization in the I2C class (the peripheral) and in main.cpp (the pins used for I2C)

https://github.com/Markiv9999/STM32NUCLEOBUS/blob/89a83315b5b10c91e3b30978a61ab5877269d3ca/Core/Src/main.cpp#L197

Markiv9999, 2024-07-11:
Checked on this, the code for pin intialization is present in step 3 of I2C Init
https://github.com/Markiv9999/STM32NUCLEOBUS/blob/89a83315b5b10c91e3b30978a61ab5877269d3ca/Core/Src/I2C.cpp#L199.
I did not notice the code was also in main, meaning the pins were getting initialized twice. Removed the I2C pin initiliazation from main.

Felt like it would help to have a seperate file for the init functions. BoardInit.cpp is now defined which calls the Hal Init, SystemClock and GPIO Init functions. We just call Board Init at the start of main. We can keep adding Board side Initializations to here as we need. 
 
Also made the UART console a self contained class similar to I2C. All its parameter and pin initialization is now inside the UartConsole Class.
This helps us seperate external peripheral initialization from the initialization of the board systems.

Therefore we now have a flow of initialization as boardinit --> peripheral initialization(I2C, UART, etc).

StefanoSperetta, 2024-07-16:
Sounds good!

closed 2024-07-16 by StefanoSperetta