Markiv9999/STM32NUCLEOBUS #25: I2C class Init not called

StefanoSperetta, 2024-07-09:
I do not see anywhere the I2C Initfunction being called. I would expect this function is needed and it should be called before initializing the TMP100 sensor.

Markiv9999, 2024-07-11:
Had setup the initialization to occur automatically the first time the constructor is called. Now have made it explicit, which is better for the flow, but feels like people can forget to initalize the object. Added the initalization code to main.

StefanoSperetta, 2024-07-16:
The big problem here is that you have 2 actions that are fundamental:
- allocation (done by the constructor)
- Initialization (done by the init function)

You typically want the constructor to be global (so visible outside the main function, not as done here https://github.com/Markiv9999/STM32NUCLEOBUS/blob/956a0ed4ad7a67eba64e376cc9e3f250e604bbc7/Core/Src/main.cpp#L46) to make sure all functions that need to access the I2C object can do it (think about event handlers, for example). But, when allocating the I2C object outside the main function, the different board peripherals have not been initialized yet when the constructor is run, this makes your code problematic. This is why it is usually good practice to separate the two actions. And if the user forgets to run the initialization, the peripheral will not work.. so it is easy to detect. The point is this is very much repeatable. If the initialization is carried out inside the constructor, the code might work if you change the order of allocation...

Anyway, closing it!

closed 2024-07-16 by StefanoSperetta